flexivrobotics/flexiv_rdk #22: Pinning a lower version of GLIBC and GLIBCXX for Python 3.10 package

smeng9, 2023-05-08:
Hi Flexiv team,

If we are trying to use Python 3.10 on Ubuntu 18 with [flexivrdk.cpython-310-x86_64-linux-gnu.so](https://github.com/flexivrobotics/flexiv_rdk/blob/main/lib_py/flexivrdk.cpython-310-x86_64-linux-gnu.so), we are likely to encounter glibc version symbol missing issue since the maximum version of glibc on Ubuntu 18 is [2.27](https://www.ubuntuupdates.org/package/core/bionic/main/updates/glibc) .

Is it necessary to use such a high version of GLIBC? Can we lower this? The compatibility table on README.md suggests Ubuntu 18 with Python 3.10 is supported.

pzhu-flexiv, 2023-06-16:
> Hi Flexiv team,
>
> If we are trying to use Python 3.10 on Ubuntu 18 with [...]

Hi, have you tried running the 3.10 lib on Ubuntu 20.04? Since GitHub pipeline has officially stopped the support for Ubuntu 18.04, we will also remove the support for Ubuntu 18.04.

smeng9, 2023-06-18:
Since 18.04 is deprecated, I believe this issue can be closed now.

closed 2023-06-18 by smeng9