flexivrobotics/flexiv_rdk #27: [FEATURE] synchronous reading of multiple control box digital inputs

conhugh, 2023-08-01:
**Is your feature request related to a problem? Please describe.**
Some users have external devices connected to multiple digital input pins on the control box, which they are hoping to read in real-time at 1kHz using RDK. However, users cannot read from multiple GPIOs in a single call, they must be read sequentially, and this does not happen fast enough to keep up with 1kHz loop. 

**Describe the solution you'd like**
Would like to be able to read from all the control box GPIO pins in a single call, at 1kHz. 

**Describe alternatives you've considered**
If the sequential reading of GPIO pins could be made faster, perhaps this could also allow for reading all the pins within a 1kHz cycle, but it would still be preferable to read all pins synchronously for applications where precise data synchronization is important.

**Additional context**
N/A.

pzhu-flexiv, 2023-08-01:
Will be added in RDK v0.8.1

pzhu-flexiv, 2023-11-29:
Fulfilled in release #40.

closed 2023-11-29 by pzhu-flexiv