flexivrobotics/flexiv_rdk #29: [BUG]

acf986, 2023-09-11:
**Version information**
- RDK: v0.8
- Robot software: v2.10.5
- OS: Ubuntu 20.04

**Describe the bug**
Start the program implementing my own controller using torque streaming API. The joint 6 suddenly swings down. While the RDK prints out:
[2023-09-07 14:52:00.937] [error] periodicTask: Fault occurred on robot server, exiting ...
[2023-09-07 14:52:00.937] [info] [flexiv::Scheduler] Stopped all tasks
[2023-09-07 14:52:01.937] [info] [flexiv::Robot] Network service closed
[2023-09-07 14:52:01.938] [info] >>>>> Flexiv RDK exited <<<<<

**Steps to reproduce**
Cannot be reproduced easily.

**Expected behavior**
The arm will stay at its initial location, and will not move.

**Screenshots**
NA

**Additional context**
Please tell us a method to find out what fault is on the server side.

pzhu-flexiv, 2023-09-11:
@acf986 Thanks for bring up the issue to us. In the upcoming (late September) v0.9 release, the server-side error message will be automatically transmitted to RDK client so you know exactly what error happened on the robot server.

acf986, 2023-09-13:
Hi, for now, is there any way we can see the possible error from you QNX system? or at least the possible type of errors that will print out this message.

pzhu-flexiv, 2023-09-13:
@acf986 Yes, you can always keep the UI tablet (Flexiv Elements) connected while using RDK. Once any error occurs, you will see a pop-up error message from the UI tablet. What we will do in RDK v0.9 is to also copy this pop-up error message to RDK client and get it printed in the console in case the UI tablet is not connected. See https://rdk.flexiv.com/manual/troubleshooting.html#fault-handling-and-error-messages

closed 2023-11-29 by pzhu-flexiv