flexivrobotics/flexiv_rdk #34: [FEATURE] setJointStiffness()

acf986, 2023-10-06:
**Is your feature request related to a problem? Please describe.**
Would like to introduce some mechanical compliance during the joint control mode.

**Describe the solution you'd like**
Add a setJointStiffness() API similarly to the current setCartesianStiffness()

**Describe alternatives you've considered**
NA

**Additional context**
NA

pzhu-flexiv, 2023-10-07:
@acf986 Joint position control is purely rigid joint control and cannot set stiffness. Joint torque control is using flexible joint, the stiffness can be changed to whatever values per your own joint impedance control implementation. For how to implement a joint impedance controller, please see https://github.com/flexivrobotics/flexiv_rdk/blob/main/example/intermediate2_realtime_joint_torque_control.cpp

acf986, 2023-10-07:
Thanks very much for your fast response.
The example given above requires a realtime loop to control the robot arm through the streamJointTorque interface.
However, in many situations, it is also nice to have access to a joint impedance controller through a non-realtime method (like sendJointPosition()).

pzhu-flexiv, 2023-10-07:
@acf986 I see, I will talk to our product team to evaluate this feature. Thanks for the bringing it up.

acf986, 2023-11-09:
@pzhu-flexiv, even in RealTime mode, it would be very helpful to have a joint compliance mode that does not use the torque streaming interface. 
Because when streaming torque command, the stability of the controller is easily affected by the communication delay. However, if we stream a joint angle command instead of the torque command (in other words, the joint space close loop control is done inside the QNX computer), the stability will be less affected by the delay.

pzhu-flexiv, 2024-03-13:
@acf986 The introduction of this new control mode is confirmed to be included in release v1.4, which should happen in June.

smeng9, 2024-04-18:
@pzhu-flexiv Will this also be available as a primitive? So we can perform different joint stiffness level in primitive commands like `MoveJComp` ?

pzhu-flexiv, 2024-04-18:
@smeng9 For now there's no plan for a joint impedance primitive. You need to use the RDK API instead.

gohlaysiong, 2024-05-07:
[image]

the python example run into error  on "import flexivsdk " using visual studio code on window pc.

pzhu-flexiv, 2024-05-09:
@gohlaysiong Please open a new issue report and post the complete error log.

closed 2024-07-15 by pzhu-flexiv