flexivrobotics/flexiv_rdk #35: [FEATURE] Singularity handling

acf986, 2023-10-06:
**Is your feature request related to a problem? Please describe.**
Currently, when the robot goes near to the singularity, it will suddenly move very fast before the emergency stop, which is quite dangerous. Like shown here: [https://youtube.com/shorts/u4vjCwGBatI?feature=share](url)

**Describe the solution you'd like**
There should be a singularity avoidance mechanism (that can be enabled or disabled by the user) so that the robot arm moves smoothly cross the singularity. Like shown here: [https://youtube.com/shorts/XGAB11z7pss?feature=share](url)

**Describe alternatives you've considered**
It shall at least stop without reaching at high speed move.

**Additional context**
NA

pzhu-flexiv, 2023-10-07:
@acf986 Please kindly provide your robot software version.

acf986, 2023-10-09:
Here are the version information:

```
[2023-10-09 19:16:39.556] [info] ================================================
[2023-10-09 19:16:39.556] [info] ===              Flexiv RDK 0.8              ===
[2023-10-09 19:16:39.556] [info] ================================================
[2023-10-09 19:16:39.557] [info] [flexiv::Robot] Network service started [RT:99]
[2023-10-09 19:16:40.57] [info] [flexiv::Robot] Connection established
[2023-10-09 19:16:40.59] [info] [flexiv::Robot] Server information:
[2023-10-09 19:16:40.59] [info] Serial number: Rizon 4s-062060
[2023-10-09 19:16:40.59] [info] Degrees of freedom: 7
[2023-10-09 19:16:40.59] [info] Software version: v2.10.5
[2023-10-09 19:16:40.63] [info] [flexiv::Robot] Initialization complete
```

pzhu-flexiv, 2023-10-11:
@acf986 This bug has been solved in robot software v2.10.7, which will be soon released with that works with RDK v0.9

closed 2023-11-29 by pzhu-flexiv