flexivrobotics/flexiv_rdk #39: [BUG] Gravity compensation cannot turn off

acf986, 2023-10-30:
**Version information**
- RDK: v0.8
- Robot software: v2.10.5
- OS: Ubuntu 20.04 - x86_64

**Describe the bug**
streamJointTorque(tau, false) does not turn off gravity compensation

**Steps to reproduce**
1. Set the gravity compensation flag of streamJointTorque to false.
2. Run a simple controller like tau = -10 * q_dot;

**Expected behavior**
The robot arm will drop because of the lack of gravity compensation.

**Additional context**
We have tried turn the gravity compensation flag on/off, but it seems there is no effect on the robot arm control at all.

pzhu-flexiv, 2023-10-30:
@acf986 This is a known issue and has been fixed in the upcoming RDK v0.9 release.

pzhu-flexiv, 2023-11-29:
Fixed with release #40.

closed 2023-11-29 by pzhu-flexiv