flexivrobotics/flexiv_rdk #42: Arm unsafe movement in in NRT motion force mode.

acf986, 2023-12-05:
**Version information**
- RDK: v0.9
- Robot software: v2.11.3
- OS: Ubuntu 20.04 

**Describe the bug**
The robot arm move in an unsafe and violent way when executing specific trajectory in NRT motion force mode.
The same issue as described in this [issue](https://github.com/flexivrobotics/flexiv_rdk/issues/35).

**Steps to reproduce**
1. Boot the robot into NRT MOTION FORCE mode.
2. Execute the reference in this file
[problem_traj.txt](https://github.com/flexivrobotics/flexiv_rdk/files/13561086/problem_traj.txt)
using sendCartesianMotionForce() api. Each line of this file is a reference pose of format [x, y, z, w, x, y, z]. And the interval between two reference pose is 20ms (the reference is to be updated at 50Hz).
3. The arm will move very violently. Sometimes triggering emergency stop.

**Expected behavior**
The arm moves smoothly.

**Screenshots**
NA

**Additional context**
Suspect there's a bug in the reference generator. The same trajectory can be executed by RT MOTION FORCE mode without any problem. (You can up-sample the reference from 50Hz to 1000Hz with linear interpolation and give it a try.)

pzhu-flexiv, 2023-12-05:
@acf986 Does the issue occur also for other update frequencies? For example, 1Hz, 10Hz and 100Hz.

acf986, 2023-12-06:
@pzhu-flexiv As long as the robot is in the NRT_CARTESIAN_MOTION_FORCE mode, it will happen. However, the same trajectory (up sampled) in RT_CARTESIAN_MOTION_FORCE mode runs fine.

pzhu-flexiv, 2023-12-06:
@acf986 Thanks for the information, we have repeated the bug and will look into it

acf986, 2023-12-15:
May I know is there any update on this issue? Or the possibility of any walk around?

pzhu-flexiv, 2023-12-15:
@acf986 We found the issue, but it's not from the RDK client side, but in the robot controller. Please wait for a hotfix.

closed 2024-03-13 by acf986