flexivrobotics/flexiv_rdk #46: [COMPILE]Request for Compatibility with Older MSVC Versions

BreadAndSalt, 2024-03-08:
I'm encountering compatibility issues with the current verison of the library on MSVC2017.It seems that the library iwas compiled with the latest MSVC version,making it incompatible with older versions like MSVC 2017,In order to use the library with MSVC2017,could you please provide a version compiled with an older MSVC toolset,such as 2017?This would greatly facilitate compatibility for users like myself who are constrained to older toolsets.Thank you for considering this request

pzhu-flexiv, 2024-03-11:
@BreadAndSalt Thanks for the feedback. According to https://learn.microsoft.com/en-us/cpp/porting/binary-compat-2015-2017?view=msvc-170, we will start to use MSVC 2017 to generate Windows libraries for future releases. This should make our library compatible with MSVC 2017 and later versions.

If you are currently in need of a MSVC 2017 version of RDK v0.9, please let us know, we will provide it to you.

BreadAndSalt, 2024-03-16:
@pzhu-flexiv  Thanks for the support! Yes , I am currently in need of a MSVC 2017 version of RDK v0.9, would you please get one for me.
Much appreciate!

closed 2024-03-19 by BreadAndSalt