flexivrobotics/flexiv_rdk #47: [BUG] Crash during primitive execution

acf986, 2024-03-12:
**Version information**
- RDK: 0.9
- Robot software: unknown (we are using your hotfix firmware)
- OS: Ubuntu 20.04 - x86_64

**Describe the bug**
Calling the following line causes the program to crash:
`flexiv::utility::parsePtStates(m_se->get_robot()->getPrimitiveStates(), "reachedTarget") == "1"`
This is line I copied from the official execute_primitive example.

Here the m_se->get_robot()->getPrimitiveStates() function call is a wrapper around the flexiv rdk's api:
```
  virtual std::vector<std::string> getPrimitiveStates() final
  {
    return robot_->getPrimitiveStates();
  }
```

**Steps to reproduce**
1. Execute a primitive that is long enough (with many waypoints)
2. There is a chance the error will happen, but need to try multiple times.

**Expected behavior**
The code does not crash.

**Screenshots**
I have valgrind my code and find the issue to be:
```
==102234== Invalid read of size 8
==102234==    at 0x4DE4804: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::size() const (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28)
==102234==    by 0x487D44: __gnu_cxx::__enable_if<std::__is_char<char>::__value, bool>::__type std::operator==<char>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (basic_string.h:6154)
==102234==    by 0x487A22: flexiv::utility::parsePtStates(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (Utility.hpp:173)
==102234==    by 0x4833BD: CompPrimState::is_finished() (comp_prim_state.cpp:48)
```
Which leads to the function 
`flexiv::utility::parsePtStates`
and the line 173 in Utility.hpp
`parsedState.front() == parseTarget`
It seems in this case the parsedState is a empty vector.

pzhu-flexiv, 2024-03-12:
@acf986 Hi, this is a known bug and has been [fixed](https://github.com/flexivrobotics/flexiv_rdk/commit/4e0ffd74a7942838e16027c71d183798a583799f) in release v0.10, which will be released soon.

acf986, 2024-03-13:
@pzhu-flexiv Sorry you mean v 1.00?

acf986, 2024-03-13:
I am sorry, in the mean while, can you suggest any quick fix that we can apply ourself to safely avoid this issue?

closed 2024-03-13 by acf986

pzhu-flexiv, 2024-03-13:
@acf986 A simple workaround is to get again if the ptStates contains corrupted element.
```
auto ptStates = robot.getPrimitiveStates();
for (const auto& v : ptStates) {
    if (v.empty()) {
        // ptStates contains corrupted data, try again
        ptStates = robot.getPrimitiveStates();
        break;
    }
}
```