flexivrobotics/flexiv_rdk #51: Where can I find API doc for different version

Valen-C12, 2024-04-26:
The [api doc](https://rdk.flexiv.com/api/index.html) is for version 0.9, but there is other newer version like 1.3, 1.4-dev in the releases, where can i find the docs for them?
BTW, what is the difference of pre-release, can i use such versions? when will them be released? after all the released is 0.9 before  almost 4 monthes, but the pre-release is already 1.4-dev :)

pzhu-flexiv, 2024-04-26:
@Valen-C12 Hi, the API doc is auto-generated by Doxygen, you can easily generate the API doc for any version by running:
```
sudo apt install doxygen-latex graphviz
cd flexiv_rdk
doxygen doc/Doxyfile.in
```
Then under flexiv_rdk/doc/html, you'll find the generated API doc. Open any .html file with your browser then you'll see the doc.

Valen-C12, 2024-04-30:
@pzhu-flexiv Got it, thank you for the quick response!