flexivrobotics/flexiv_rdk #55: Allow Joint Freedrive Floating from RDK

joepuzzo, 2024-06-01:
Currently I can enable freedrive floating via

```js
executePrimitive(
            f"FloatingCartesian(cartFloatingAxis={floatingAxis}, cartFloatingFrame={cartFloatingFrame}, enableJointFloating={enableJointFloating})"
        )
```

But I would really also love the ability to enable freedrive floating for individual joints as well. Is there a `FloatingJoint` function hidden away somewhere ? :)

pzhu-flexiv, 2024-06-03:
@joepuzzo What software version are you using?

joepuzzo, 2024-06-03:
For RDK im using 0.10

pzhu-flexiv, 2024-06-03:
@joepuzzo Try primitive `floatingSoft`. It has the following parameters:
- `jointFloatingSwitch`: switch for each joint floating, 0 for off, 1 for on. Data type: Vector7d. Default value: 1.0 1.0 1.0 1.0 1.0 1.0 1.0
- `dampScale`: scale of joint damping. Data type: double. Range: 1.0 - 10.0. Default value: 1.0.

joepuzzo, 2024-06-04:
[image]

Seems that does not exist

pzhu-flexiv, 2024-06-04:
@joepuzzo It's floatingSoft not FloatingSoft

joepuzzo, 2024-06-04:
Oh wow I just guessed because "FloatingCartesian" is uppercase that `floatingSoft` would also be uppercase haha

joepuzzo, 2024-06-04:
Will give it a try now

joepuzzo, 2024-06-04:
That worked! Thanks!

closed 2024-06-04 by pzhu-flexiv

yellownebula10, 2024-07-01:
@pzhu-flexiv I would like to use the primitives `FloatingCartesian` and `floatingSoft` in my application, but they do not seem to be mentioned in the [primitives documentation](https://original.flexiv.com/en/primitives/index.html#flexiv-primitives-manual). Where can I find documentation for these primitives?

kaibiao-flexiv, 2024-07-01:
Hi @yellownebula10, `FloatingCartesian` and `floatingSoft` are not standard primitives, and therefore not documented on our primitives page. If you are interested, you can import [PLAN-FreeDriveAuto.zip](https://github.com/user-attachments/files/16059135/PLAN-FreeDriveAuto.zip) on Elements and check out these primitives' parameters and descriptions.