flexivrobotics/flexiv_rdk #56: Ability to Jog Robot via RDK

joepuzzo, 2024-06-01:
I would love if there were a a primitive that would allow me to jog the robot. Currently I can achieve this by looking at current cart pos or joint pos and only modify the field I want, however it would be pretty sweet if this was a built in function.

pzhu-flexiv, 2024-06-03:
@joepuzzo Joint jogging can be easily achieved using `NRT_JOINT_POSITION` control mode.

joepuzzo, 2024-06-04:
IS there example of executing jog via this control mode in RDK ?

pzhu-flexiv, 2024-06-04:
@pzhu-flexiv Not on joint jogging specifically, but [this example](https://github.com/flexivrobotics/flexiv_rdk/blob/main/example_py/intermediate1_non_realtime_joint_position_control.py) should give you the idea.

closed 2024-06-10 by pzhu-flexiv