flexivrobotics/flexiv_rdk #59: Unable to connect/CommException[BUG]

jspinks8144, 2024-06-10:
**Version information**
- RDK: v0.10
- Robot software: v2.10.4
- OS: Windows 10 Pro, v22H2, x64, Visual Studios 2022 and Visual Studios 2019

**Describe the bug**
Sample programs crash immediately. 

Attempting to run any sample program in Visual studios with IP addresses hardcoded throws a CommException.

**Steps to reproduce**
1. install rdk, enable rdk remote on robot server, boot in auto mode, connect to robot server over ethernet.
2. Windows 10, run cmd:
'start basics1_display_robot_states.exe 192.168.2.100 192.168.2.xxx'
 using git bash or cmd prompt while in the rdk/example/build/release dir.

**Expected behavior**
For program to connect to robot server.

**Screenshots**
CodeSnippet: Shows the line that throws the exception. see the comment.
[image]
Exception: The exception that is thrown.
[image]

 I'm able to ping the robot server.

kaibiao-flexiv, 2024-06-10:
Thank you for providing the detailed information, it looks like there is a compatibility issue. The latest RDK version(v0.10) is only compatible with the Flexiv software package at v2.11.5.

Please clone/download RDK tag version at v0.7 (https://github.com/flexivrobotics/flexiv_rdk/releases/tag/v0.7), which is compatible with your current Flexiv software package version (v2.10.4). 

FYI, the RDK history releases and their respectively compatible Flexiv software package versions are listed here at: https://github.com/flexivrobotics/flexiv_rdk/releases. 

Additionally, Flexiv software package version information can be found on Flexiv Element (Settings -> Version & Update).

Thanks!

KaibiaoRuan, 2024-06-10:
Thank you for providing the detailed information, it looks like there is a compatibility issue. The latest RDK version(v0.10) is only compatible with the Flexiv software package at v2.11.5.

Please clone/download RDK tag version at v0.7 (https://github.com/flexivrobotics/flexiv_rdk/releases/tag/v0.7), which is compatible with your current Flexiv software package version (v2.10.4).

FYI, the RDK history releases and their respectively compatible Flexiv software package versions are listed here at: https://github.com/flexivrobotics/flexiv_rdk/releases.

Additionally, Flexiv software package version information can be found on Flexiv Element (Settings -> Version & Update).

Thanks!

closed 2024-06-11 by jspinks8144