flexivrobotics/flexiv_rdk #60: [FEATURE] Robot Controller Network and DHCP configuration

fvcouture, 2024-06-17:
I am unable to find how to change the ip address of the robot controller
192.168.2.100

I would like to be able to change the ip address of the controller and also be able to enable or disable the DHCP server on the controller (Looks like the DHCP Server ip address is 192.168.2.1)

ideally we want to set a static address on the interface,
if the controller works as a dhcp client (rather than dhcp server) that works too.

We want to be able to reach multiple controllers on our network and we need to be able to control their IP and network range (i.e. not be limited to the 192.168.2.0/24 network)

kaibiao-flexiv, 2024-06-17:
Hi @fvcouture, please reference the information below

## Connect to Router Settings
1. Connect a PC to the control box's **LAN** port
[image]
2. Open a browser
3. Go to address 192.168.2.1
     - user name: adm
     - password: 123456
[image]
4. If the language is not English, go to "Settings" and set the language to English

## Modify IP Configurations
### Scenario A: Change IP for _a single robot_ within its own router network (only 192.168.2.xxx)
1. Navigate to Service" → "DHCP Service" page
[image]
2. Change the IP Pool Starting Address and IP Pool Ending Address to include your new address
     - For example, if you want to change the RobotPC IP to 192.168.2.200, set "start" to 192.168.2.200, and "end" to 192.168.2.210. 
     - **Note:** If modifying the DHCP address, we need to ensure that the DHCP address and IP Pool 192.168.2.200 - 192.168.2.210 are NOT overlapped. And please keep the new IP address under 192.168.2... and also do not use 192.168.2.1, because the router IP is 192.168.2.1
[image]
3. Change the Static DHCP address of Host/RobotPC
     - For example, if you would like to change the RobotPC IP to 192.168.2.200, set the "IP Address" of RobotPC to 192.168.2.200 (make sure the IP is within the DHCP IP Pool)
     - **Note:** Please remember to click “add“ after one assignment.
[image]
4. Apply the changes and reboot

### Scenario B: Change IP(s) for _a group of robots_ to work under one router network
1. **Note:** follow this step for scenarios such as establishing a network among robots [under 192.168.2.xxx only] or using one off-shelf router to establish a network among robot(s) [under any xxx.xxx.xxx.xxx depending on the router]
2. For **slave robot(s)**, navigate to Service" → "DHCP Service" page
     - (*important do not delete) Note down RobotPC's MAC Address
     - Uncheck the "Enable DHCP" in the “Static DHCP assignments“ section
     - Apply the changes
3. For the **master robot/external router**, navigate to the Service" → "DHCP Service" page
     - Add the noted slave robot(s)' MAC Address to the list of Static DHCP
     - For example, to create a network of 2 robots. 192.168.2.100 remains the same master robot and another 192.168.2.110 for the other slave robot (if more than two robots, set one robot/router as the master, and others as slaves).
[image]
     - Apply the changes
4. Reboot the master/router and slave robots to take effect

## Update Robot IP Configuration on Tablet Elements if necessary
1. Open ConfigIP
[image]
2. Change line 4 IP to the new address
[image]
3. Ping the new address to confirm the connection
[image]

fvcouture, 2024-06-18:
Hello Kaibiao

Thank you so much for your response! This was very helpful.
What I was missing is that I can access the web ui of 192.168.2.1
(are those instructions anywhere in the documentation? Should I have read some docs instead of asking here?)

However, the hints on that post is not exactly what I wanted to do,

Here's what I did:

- Disable the DHCP server on 192.168.2.1
- Change the VLAN1 interface address to 10.10.10.2

on that 10.10.10.XXX network:
- I already have an existing gateway 10.10.10.1
- I already have an existing DHCP server 

So know the controller is in static address for VLAN1, once that configuration is in, the robot is getting a 10.10.10.XXX IP from my dhcp server

Ideally I would have loved to be able to configure the robot IP statically as well, but its ok as I can do a DHCP Reservation on my dhcp server (similar to what was done on the controller)

And I still have a routing issue. The controller and robot are reachable from the 10.10.10. network, but they are not using my gateway (10.10.10.1) to get out of that subnet and be able to ping/connect to other devices

i.e. My workstation is on network 10.10.11.X, I can ping all devices on 10.10.10.X except the robot and controller
When I connect my workstation to network 10.10.10.X, I can ping the robot and controller, hence why I think they are not using 10.10.10.1 as gateway to reach other subnets.

on the controller admin web page, in Network -> Static Routes
I did change the default 0.0.0.0 route to use 10.10.10.1 as gateway

Apologies for all the technical details here, my 2 questions are:

1. Can we solve the routing issue on the controller (when the controller is not the actual router but just a network client)

question 2 is much lower priority to us:
2. Is it possible to assign a static IP to the robot, it seems to always request a DHCP lease

kaibiao-flexiv, 2024-06-20:
Hi @fvcouture,

Thank you for explaining further! Since making the controller a network client and assigning static IP involves more lower-level configurations, please send me a direct email (in my profile) and we can schedule a time to complete these configurations on a meeting call. Thanks!

fvcouture, 2024-06-21:
Hi kaibiao!

I will send you an email next week and we can proceed from there

Many thanks!

closed 2024-07-08 by pzhu-flexiv