flexivrobotics/flexiv_rdk #61: [BUG] get global var of COORD ARRAY type via rdk will be truncated at 127 length

Valen-C12, 2024-06-24:
**Version information**
- RDK: 0.10
- Robot software: v2.11.5
- OS: Ubuntu 20.04 - x86_64

**Describe the bug**
Look at the 'contact_points', the element shows it contain 7 coord points, but in value get via rdk, the value is truncated

[image]
[image]

**Steps to reproduce**
1. store a global var of coord array type with more than 2 points via element ui
2. use rdk to get the global vars
3. See error

**Expected behavior**
The global vars get via rdk should be full and no any truncated

pzhu-flexiv, 2024-06-25:
@Valen-C12 Hello, this bug no longer exists in RDK v1.4, which is expected to be released early next month. Below shows getting and printing 2 global variables. The first contains 7 ARRAY_COORD points, the second contains 20  ARRAY_COORD points. Both are complete.
[image]

closed 2024-06-25 by pzhu-flexiv

Valen-C12, 2024-06-26:
@pzhu-flexiv Nice to hear that, looking forward to the new version