flexivrobotics/flexiv_rdk #62: Add self-examination and calibration routines to RDK

yellownebula10, 2024-06-25:
**Is your feature request related to a problem? Please describe.**
It would be helpful to have the ability to trigger self-examination and calibration routines remotely, without using the UI. 

**Describe the solution you'd like**
Add the self-examination and calibration routines to the RDK.

**Describe alternatives you've considered**
I tried to run the joint performance test by remotely executing a plan. 
The plan execution runs fine but there is no option to get/save/examine data nor apply it to the robot. 
Adding a PASS/FAIL test result would be helpful (similarly to what done when executing from the Elements UI).

pzhu-flexiv, 2024-07-09:
@yellownebula10 This feature will be added in RDK v1.5.

yellownebula10, 2024-07-10:
That's great news, thank you @pzhu-flexiv! Do you know when RDK v1.5 will be released? Will this feature allow to get the pass/fail values from audit and to apply calibration values from the RDK?

pzhu-flexiv, 2024-09-03:
@yellownebula10 Due to limited bandwidth, this feature will have to be postponed to v1.6. Thanks for your understanding.

yellownebula10, 2024-11-23:
Hi @pzhu-flexiv, 
what is the timeline for RDK v1.6? Will it be compatible with SW v3.7?

pzhu-flexiv, 2024-11-26:
RDK v1.6 will be released with SW v3.8 by the end of Dec. RDK v1.5 is for SW v3.7

pzhu-flexiv, 2025-01-09:
@yellownebula10 This feature is fulfilled in release v1.6, see https://github.com/flexivrobotics/flexiv_rdk/blob/main/include/flexiv/rdk/maintenance.hpp

closed 2025-01-09 by pzhu-flexiv