flexivrobotics/flexiv_rdk #64: [HELP] streamCartesianMotion Input Validation

yellownebula10, 2024-06-28:
Under the documentation for streamCartesianMotionForce, it says to stream smooth and continuous motion commands. Is there a standard we can use to validate that a stream is smooth enough? In other words, what is the maximum allowable delta between two subsequent stream values for the position and the quaternion?

kaibiao-flexiv, 2024-06-29:
Hi @yellownebula10 ,

When using streamCartesianMotionForce, it is up to the user to create the trajectories and motion control profile for robot actions. You can refer to this picture below when streaming cartesian positions:
[image]
[Trapezoidal Velocity Profile from [Mathworks](https://www.mathworks.com/help/robotics/ug/design-a-trajectory-with-velocity-limits-using-a-trapezoidal-velocity-profile.html)]

For linear position on Rizon arms:
- The typical **peak velocity** for Rizon arms is at 1 m/s (with max velocity at 2.2 m/s, but note this max velocity may trigger robot fault in some challenging joint poses).
- The typical **peak acceleration** for Rizon arms is at 1.5 m/s^2 (with max acceleration at 3 m/s^2).

For angular rotation on Rizon arms:
- The typical **peak angular velocity** for Rizon arms is at 300 deg/s (with max angular velocity at 800 deg/s, but note this max angular velocity may trigger robot fault in some challenging joint poses).
- The typical **peak angular acceleration** for Rizon arms is at 300 deg/s^2 (with max acceleration at 800 deg/s^2).

Since Streaming is at 1kHz frequency, you could multiply the above numbers by 0.001 seconds to get the delta distance. During the **"Phase 2 constant speed"** of the motion control profile, with a typical peak velocity, the maximum allowable delta distance is 0.001[s] * 1[m/s] = 0.001[m] (This is not applicable during "Phase 1 Acceleration" and "phase 3 Deceleration").

Using RT streaming control can be quite complex. If your trajectory points are more spread out and less complex (recommended minimum 100Hz frequency), you could save some time by utilizing an NRT discrete cartesian control [sendCartesianMotionForce] and letting the robot's internal motion generator does all the difficult work.
https://github.com/flexivrobotics/flexiv_rdk/blob/06b38ad52fc93b0ffa0b4976d734e2221bf523a4/include/flexiv/Robot.hpp#L495

yellownebula10, 2024-07-01:
Thank you @kaibiao-flexiv . If we were to record the robot's position while in joint floating mode and wanted to stream it back, given that the trajectory is not complicated, would it be better to use NRT streaming at <100Hz or RT streaming at 1kHz? We are already automatically recording all the robot states at 1kHz, so it would just be a matter using every point or not.

kaibiao-flexiv, 2024-07-01:
Hi @yellownebula10 ,

If the position data was recorded at 1kHz, you should be able to RT stream them back to the robot at 1kHz in order to replay the action. If PC's RT streaming performance is not perfect at 1kHz rate, then you can downsample the data to 10% (by averaging them per 10 sample data) and apply NRT sendCartesianMotionForce at 100Hz.

yellownebula10, 2024-07-01:
Got it, thanks @kaibiao-flexiv

closed 2024-07-01 by yellownebula10