flexivrobotics/flexiv_rdk #65: [BUG] Termination condition for Contact Primitive

yellownebula10, 2024-06-28:
**Version information**
- RDK: v0.10
- Robot software: v2.10.9
- OS: macOS 14.5 - arm64

**Describe the bug**
When polling primitive states to determine that the contact primitive has terminated, if the termination condition is set to 'terminated' == '1' as specified by the primitives documentation, the condition is never met.

**Steps to reproduce**
1. Send contact primitive to robot with default parameters
2. Add a while loop polling current primitive states, checking for 'terminated' == '1'
3. See error

**Expected behavior**
The primitive state 'terminated' should be set to '1'. However, when I print the primitive states as the primitive executes, I notice that ‘terminated’ remains at ‘0’ while ‘primitiveName’ switches to ‘Hold’ from ‘Contact’ and ‘reachedTarget’ is set equal to ‘1’.

**Screenshots**
[image]

kaibiao-flexiv, 2024-06-29:
Hi @yellownebula10 ,

In our system, some primitives have a self-termination condition, which means that when the termination condition is met, the primitive will automatically terminate. The Contact primitives you mentioned are among those primitives that can self-terminate. (This is because after contact is made, the robot should not be staying within Contact primitive state)

So the reason that you were not able to reach the 'terminated' == '1'' condition is because at the time you query, the primitive has already met the self-termination condition and has already exited, and the primitive states are no longer available as that primitive is no longer running.

To identify which primitives will terminate by itself: 
If you spot the primitive's default transition condition is [terminated == 1], then this means it is one of the self-terminated primitives [Flexiv Primitives default transition condition](https://original.flexiv.com/en/primitives/Basic%20Force%20Control.html?#default-transition-condition)
[image]

To check if such primitives are terminated or not, you can use the isBusy() method, below is an example code snippet:
[image]

yellownebula10, 2024-07-01:
Thank you @kaibiao-flexiv !

closed 2024-07-01 by yellownebula10