flexivrobotics/flexiv_rdk #66: [FEATURE] Get/Set WorkCoord(工件坐标系) via RDK

Valen-C12, 2024-07-04:
**Is your feature request related to a problem? Please describe.**
使用element ui 设定比如moveL的目标点的时候，可以选定变量作为坐标系，但示教只能选世界/法兰/工具坐标系。
[image]
[image]

**Describe the solution you'd like**
rdk能够更新工具坐标系，或者示教可以选择COORD变量作为坐标系

**Additional context**
场景是机械臂是在移动平台上的，每次到位后机械臂和操作工件可能会有几厘米误差，所以需要校准。校准好后把工件坐标系通过rdk更新到机械臂，目前只能写入到全局变量中。
因为拖拽示教只能基于工件坐标系不能基于全局变量，所以录制动作/plan的时候都得先校准一遍，然后人手动在element里把全局变量中的坐标更新到工件坐标系里，然后基于工件坐标系录动作，体验很不好。

pzhu-flexiv, 2024-07-08:
@Valen-C12 Editing workpiece coordinate will be added in RDK v1.5

Valen-C12, 2024-07-09:
> @Valen-C12 Interacting with workpiece coordinate will be added in RDK v1.5

Cool, when will v1.5 be released? @pzhu-flexiv

pzhu-flexiv, 2024-07-09:
@Valen-C12 It is scheduled to be released late Q3 / early Q4.

pzhu-flexiv, 2024-11-12:
https://github.com/flexivrobotics/flexiv_rdk/blob/main/include/flexiv/rdk/work_coord.hpp

closed 2024-11-12 by pzhu-flexiv