flexivrobotics/flexiv_rdk #67: [FEATURE] Read safety inputs

isa-9, 2024-07-04:
RDK version: 0.10
**Describe the solution you'd like**
I would like to be able to read the safety inputs via RDK. Currently, I'm only able to read digital inputs

pzhu-flexiv, 2024-07-08:
@isa-9 Could you elaborate how you plan to use safety inputs?

isa-9, 2024-07-10:
@pzhu-flexiv I would read their values to handle correctly in my code the cases where the robot is in emergency stop (from external e-stop), when it's in reduced state or when its safeguard is not right

pzhu-flexiv, 2024-07-15:
@isa-9 We cannot expose raw safety I/O to the users, however, RDK v1.5 will add a function to get if the robot is in reduced state or not.

isa-9, 2024-07-17:
In that case, would it be possible to know why the Robot::isOperational would return false i.e. would it be possible to get the reason not only by printing it on the console like it is done currently with the verbose argument, but by returning that information, for example. That way, we could know if the robot is not operational because one of these reasons and treat that information in the code:
- The robot is not enabled
- The brakes are not fully released
- The mode of the robot is auto (not remote)
- The mode of the robot is manual
- The robot is in fault
- The robot is in reduced state
- The safeguard is not ok
- The external e-stop is pressed

pzhu-flexiv, 2024-07-17:
@isa-9 Sure, I can convert the not operational reasons to an enum in RDK v1.5.

closed 2024-11-12 by pzhu-flexiv