flexivrobotics/flexiv_rdk #69: [FEATURE] Catch exception by type in pyhton

skyrimax, 2024-07-08:
**Is your feature request related to a problem? Please describe.**
I cannot catch exceptions by type in python, python's generic RuntimeError is thrown instead.

**Describe the solution you'd like**
I would like to be able to catch exceptions by the types defined in the C++ RDK, like this:

```
import flexivrdk

robot_ip = '192.168.1.100
local_ip = '192.168.1.101

try:
    flexivrdk.robot(robot_ip, local_ip)
except flexivrdk.InitException as e:
    # do some stuff
except flexivrdk.CompatibilityException:
    exit(1)
except flexivrdk.CommException:
    # do some other stuff
```

**Describe alternatives you've considered**
I thought about parsing the exception's message string, but that required access to the source code to make sure all cases are covered, but I assume that is not possible

**Additional context**
I am using python 3.10 on both x86_64 as well as arm64 machines
I installed the corresponding .so libraries in /usr/local/lib/python3.10/site-packages/ on both machines

pzhu-flexiv, 2024-07-08:
@skyrimax You can do this in RDK v1.4, which will be released by the end of this week. The mapping between RDK exceptions and Python exceptions can be found [here](https://pybind11.readthedocs.io/en/stable/advanced/exceptions.html#built-in-c-to-python-exception-translation).

pzhu-flexiv, 2024-07-08:
@skyrimax Yes, you will have to upgrade to Flexiv Software Package v3.6. You can contact your sales rep and request the upgrade.

skyrimax, 2024-07-09:
Ok, will do, thank you

skyrimax, 2024-09-18:
Hello,

Sorry to reopen this thread. We have update our Rizon 10 to use RDK 1.4 and I noticed that most standard C++ exceptions thrown by the RDK collapse into RuntimeError, so I still have to distinguish them by string comparisson. I was wondering if that was taken into account.

pzhu-flexiv, 2024-09-23:
@skyrimax Could you provide an example where programmatically accessing the thrown exception is a must-have?