flexivrobotics/flexiv_rdk #70: [BUG]

Valen-C12, 2024-07-09:
**Version information**
- RDK: [v0.10]
- Robot software: [v2.11.5]
- OS: [Ubuntu 20.04 - x86_64]

**Describe the bug**
We use the rdk to set a global variable(https://rdk.flexiv.com/api/classflexiv_1_1_robot.html#a192bccc4dec29e42755114740eace6a7), and the global variable was set successfully, checked from the element UI. But after about one minute, the global variable was changed to previous value for no reason.

我们一开始用RDK设了一个global variable，在pad上看是改成功了，后来过1分钟发现又变回原来的值了，这个可能是什么原因呢？

**Steps to reproduce**
1. update one globel variable via rdk
2. check it is updated successfully in the element ui, from the plan-variable tab
3. do some operations via the element, like update one workpiece coordinate
4. the global variable is reset to a different value

**Expected behavior**
the global variable should not be updated silently for no reason, and should be updated to the latest one in robot and the element

**Additional context**
We connect 2 robot arms from one host computer, both arms use static ip that updated by the flexiv engineer.
Both arms have this issue

pzhu-flexiv, 2024-07-09:
@Valen-C12 When you say the global variable was changed back, did you see the value changed back from Elements or from RDK?

Valen-C12, 2024-07-10:
> @Valen-C12 When you say the global variable was changed back, did you see the value changed back from Elements or from RDK?

from Element. After see it changed back I have chacked the data sync in the main page, it seems the robot is the new data but the element one is old. So when we play plan that using the global value from element, there will be problem.

I didn't check the value using rdk but i think it should be the same one as the `robot` value seen in the element.

[image]

pzhu-flexiv, 2024-07-10:
@Valen-C12 Could you double check that getGlobalVar from RDK also returns the wrong values?

Valen-C12, 2024-07-10:
> @Valen-C12 Could you double check that getGlobalVar from RDK also returns the wrong values?

I have checked using rdk, it return the right, that is the updated values

pzhu-flexiv, 2024-07-10:
@Valen-C12 It seems like an Flexiv Elements display issue, I'll report this to our team. Thanks!

Valen-C12, 2024-07-11:
> @Valen-C12 It seems like an Flexiv Elements display issue, I'll report this to our team. Thanks!

Not only display issue, it will also effect the plan editing & go action using the elements.
For example we have one moveL action that targeting one position under the global variable `table` coord.
After update the `table` using rdk, we run this moveL, it will goto the right position, and the elements display the right value.
But after around one minute without doing any thing, we run this action, it goto the wrong position. And then we check the global value in elements, it changed. That is also how we found this issue

[image]

Valen-C12, 2024-07-11:
@pzhu-flexiv 

We record one vedio to reproduce
1. comfirm the global `table` is `0 0 0 0 0 0` using rdk and elements
2. update the `table` to 1 2 3 4 5 6 using rdk
3. comfirm the global `table` is changed to `1 2 3 4 5 6` using rdk and elements
4. update one work coord using element, don't need truely updating, only edit and save without midify is okey
5. check the global `table` using rdk and elements, it is `0 0 0 0 0 0` now

https://github.com/flexivrobotics/flexiv_rdk/assets/123928892/6afe0750-a58a-4c9a-a17c-e7b377748b71

pzhu-flexiv, 2024-07-15:
@Valen-C12 This issue no longer exists in the just-released RDK v1.4, please give it another try.

closed 2024-07-15 by pzhu-flexiv