flexivrobotics/flexiv_rdk #73: [BUG] Mating primitives not working

JoeProg-ets, 2024-07-15:
- RDK: 0.10
- Robot software: v2.10.9
- Rizon 10

The Mating primitives on our RIzon 10 AND Rizon 4s don't work even with the good parameters.  We tried the 3 different primitive available, and the robots never seemed to detected the desired force. We even tried to stop (by-hand) the arm, but it only triggered the safety force threshold. We tried setting up the mating force from 10N to 70N, but still the same result. They never mate.

kaibiao-flexiv, 2024-07-16:
Hi @JoeProg-ets ,

Would you mind sharing how you are using the Mating primitives?

The [Mating primitives](https://original.flexiv.com/en/primitives/Adaptive%20Assembly.html#matingslide) control the robot to complete a hybrid motion and terminates when the total number of mating times is finished (which is set by their matingTimes parameters).  Therefore, Mating primitives only apply the desired force but are not used to detect any force. Changing the matingForce parameter will only change how much force the robot is applying while moving in the mating trajectories, but it doesn't affect its matingTimes termination condition.

JoeProg-ets, 2024-07-17:
Hi @kaibiao-flexiv ,

I think that the primary fonction of the mating primitives is not doing what we would expect from it.  Ideally, it should detect opposing forces and respond accordingly, rather than simply moving past the desired alignment points. Alternatively, it would be helpful if it could return to the furthest point on the Z-axis to ensure a snug fit. 

Currently, we have male and female parts designed to mate, with the male part having three cones and the female part having three corresponding holes. Unfortunately, using the current primitives, the cones enter the holes but the robot then continues its trajectory (Slide, Rob, or Zig-Zag), causing it to exit the holes again. Despite multiple repetitions (matingTimes), the two parts do not remain mated. We've tried various parameters without success.

Could you please advise on how we might resolve this issue?

Thank you

gavin-flexiv, 2024-07-17:
Hi @JoeProg-ets ,

The mating primitive is recommended to be used after the parts have been aligned as a way to wiggle or slide during insertion.

However for your application I suggest using a search primitive to align the parts followed by a contact primitive . The search primitive will allow you to exit the primitive based on the opposing force or z-axis position like you mentioned. You can choose between a spiral and zig zag search pattern.

Then to ensure a snug fit the most simple option is to use a contact primitive. You can control how hard the robot will push the parts together with the max contact force parameter.

Let us know if these suggestions work for your application.

pzhu-flexiv, 2024-07-24:
@JoeProg-ets Is your issue solved?

pzhu-flexiv, 2024-09-03:
> @JoeProg-ets Is your issue solved?

Closed issue due to inactivity.

closed 2024-09-03 by pzhu-flexiv