flexivrobotics/flexiv_rdk #77: [BUG] getPrimitiveStates error no response from server

acf986, 2024-08-12:
**Version information**
- RDK: v0.10
- Robot software: v2.10.9
- OS: Ubuntu 20.04

**Describe the bug**
After starting a primitive from the RDK, the function getPrimitiveStates is used to monitor the primitive's executing status. However, it sometimes gives the error No response from server and crashes the program. Unlike in the previously reported issue [here](https://github.com/flexivrobotics/flexiv_rdk/issues/54), I am only calling getPrimitiveStates at 10Hz.

**Expected behavior**
It at least should not crash the program.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Will catch the thrown out error help in this case?

pzhu-flexiv, 2024-08-12:
@acf986 Have you caught the exception thrown by getPrimitiveStates() when the error "No response from server" happens? Exception will be propagated to the top of stack and trigger the program to exit if not handled. 

Server response timeout can happen under busy networks. To address this, you can write a try catch statement to handle such exception: put another getPrimitiveStates() call in the catch {} statement to retry getting primitive states if the previous failed.

acf986, 2024-08-13:
@pzhu-flexiv Thanks for the heads up. We’ve made changes and are now testing. Since this isn’t a consistent issue, testing might take a while.

We noticed a specific error message, `[2024-07-31 15:34:28.134] [info] [flexiv:: Robot] Network service closed` when the program previously crashed. This suggests the RDK might be closing the network service itself when it thinks the server is not responsing.

We’re unsure if catching the exception will prevent this. Could you confirm if that's possible?

pzhu-flexiv, 2024-08-13:
@acf986 (1/2) What are the primitives you are using?

acf986, 2024-08-13:
> @acf986 (1/2) What are the primitives you are using?

I am using the moveCompliance primitive

pzhu-flexiv, 2024-08-13:
@acf986 (2/2) Could you update the get primitive states part of your code to the following and try again?
```
std::vector<std::string> states;
do {
    try {
        states = m_se->get_robot()->getPrimitiveStates();
    } catch (const flexiv::CommException& e) {
        std::cout << "Get primitive states failed, trying again ..." << std::endl
    }
    std::this_thread::sleep_for(std::chrono::milliseconds(100));
} while (flexiv::utility::parsePtStates(states, "reachedTarget") != "1");
```

pzhu-flexiv, 2024-08-13:
> > @acf986 (1/2) What are the primitives you are using?
> 
> I am using the moveCompliance primitive.

Then you have two options, one is to try the revised code I provided above, the other is to use Robot::isStopped() with some initial wait until the robot starts to move.

pzhu-flexiv, 2024-09-03:
Closed issue due to inactivity.

closed 2024-09-03 by pzhu-flexiv