flexivrobotics/flexiv_rdk #78: [FEATURE] Set arbitrary force control direction

acf986, 2024-08-18:
**Is your feature request related to a problem? Please describe.**
The RDK's current force control capabilities, provided by SetForceControlAxis and SetForceControlFrame, are restricted to aligning with the x, y, or z axis of the WORLD or TCP frame. For instance, when using a power file to polish the interior of a circular workpiece, the TCP's orientation should remain constant, while the force control direction should dynamically adjust to match the trajectory.

**Describe the solution you'd like**
To increase flexibility, we should be able to define the force control frame using an arbitrary rotation matrix. This setting should also be dynamically adjustable during operation, allowing for a force control axis that changes in real time.

pzhu-flexiv, 2024-08-20:
@acf986 We are running a bit low on bandwidth for new features for the next release. Could you evaluate how urgent is this feature for you on the scale from 1 to 10?

pzhu-flexiv, 2024-08-28:
@acf986 We have scheduled this feature for v1.6 due to limited bandwidth.

pzhu-flexiv, 2024-09-25:
@acf986 This feature is added to v1.5 because another feature depends on it.