flexivrobotics/flexiv_rdk #79: [BUG]

acf986, 2024-09-12:
**Version information**
- RDK: v0.10
- Robot software: v2.10.9
- OS: Ubuntu 20.04 - x86_64

**Describe the bug**
Following the official example, we uses `Robot::isBusy()` function to check whether a plan has been finished.
However, we noticed if the robot is stopped by the safety IO, `isBusy()` will return false and the mission will be treated as finished sometimes.

**Steps to reproduce**
1. use `executePlan` to start a plan.
2. before the plan finishes stop the robot with safety IO.
3. repeat multiple times to see the issue.

**Expected behavior**
Is there any other function we could use to check whether the robot is stopped by the safety IO so the plan is not really finished?
Should we use `isOperational` in this case?

pzhu-flexiv, 2024-09-12:
@acf986 Which exact safety IO are you referring to?

acf986, 2024-09-13:
@pzhu-flexiv we are following the safety input/output shown here:
[image]

pzhu-flexiv, 2024-09-13:
@acf986 When safeguard stop input is high, a CAT2 stop of the robot will be triggered, which will force the robot to stop whatever it's currently doing, which of course includes a running plan. `Robot::isBusy()` returns whether the robot is doing anything, in the case of CAT2 stop, the robot is doing nothing, therefore the function is returning the correct status.

In that case, to know whether the plan has finished before the interruption, you can monitor `robot.plan_info().node_path`.

acf986, 2024-09-14:
@pzhu-flexiv Could you kindly gives a minimum example?

pzhu-flexiv, 2024-09-16:
@acf986 Sure. Say you have a plan like this and you assigned it to the robot so it's visible to RDK.
[image]

Then in your program, just check if the current node_name matches the last node that's not "End" (because "End" only persists for several ms so it's hard to catch with a low frequency while loop). If a match is found, that means your plan has reached the last node thus can be regarded as finished. Here I'm using a dummy Hold primitive with node name "Finished" for this purpose.
[image]

pzhu-flexiv, 2024-09-25:
Closing due to inactivity.

closed 2024-09-25 by pzhu-flexiv