flexivrobotics/flexiv_rdk #81: [FEATURE] Could robot.operational() return an error code instead of a simple boolean for error handling

skyrimax, 2024-09-19:
**Is your feature request related to a problem? Please describe.**
When using reobot.operational(), I only have logs to inform me on why the robot is not operational. I cannot use those to do automatic error handling nor inform an operation that uses and HMI.

**Describe the solution you'd like**
I would like the method to return error codes instead of a simple boolean, something like this:

0: no problem, robot is operation
1: robot is not enabled
2: breaks are engaged
4: Not in auto mode (remote)
8: has fault
16: in reduced state
32: e-stop engaged

I put each error on it's own bit so it can be known if there are multiple error at the same time, but I don't think that is strictly necessary.

**Describe alternatives you've considered**
I tried to bypass this limitation by using the contextlib.redirect_stdout and contextlib.redirect_stderr context mangers to capture the flexivrdk's messages in a string, but in both cases the message was printed and nothing was captured.

Another posiblity is to expose other methods to probe each problems individually. Something like:

bool Robot::enabled(); // True if robot is enabled, false if robot is not enabled
bool Robot::breaks_engaged(); // True if breaks are engaged, false if breaks are released
int Robot::auto_mode(); // 0 if manual mode, 1 if auto mode, 2 if auto mode (remote)
int Robot::fault(); // Return error code of the fault instead of a simple boolean
bool Robot::reduced_state() // True if robot is in reduced state, false if robot is not in reduce state
bool Robot::external_estop_released(); // Complementary method to give the state of the external e-stop botton that can be wired in the control box

**Additional context**
N/A

pzhu-flexiv, 2024-09-23:
@skyrimax Is this what you are looking for?
https://github.com/flexivrobotics/flexiv_rdk/blob/release/v1.5/include/flexiv/rdk/robot.hpp#L112

skyrimax, 2024-09-23:
@pzhu-flexiv Yes, that's what I'm looking for. THank you for pointing this to me, it is not listed in the RDK's documentation. Since we are developping in Python, the online documentation has been our only way to know what methods are available for us to use. Looking at the top of the page, it does say RDK 1.4 so we assumed it was up to date. Thanks for your help, I'll test this method when I have the time and will close the issue if it solves our issues. I am leaving the url of the documentation we used as proof that this method is not listed.

https://rdk.flexiv.com/api/classflexiv_1_1rdk_1_1_robot.html

pzhu-flexiv, 2024-09-23:
@skyrimax The link I provided is in release/v1.5 branch, which is not released yet. If you are watching the repo, you should receive a notification when v1.5 is released. At that time the online documentations will be updated as well. Thanks for your feedback!

closed 2024-11-12 by pzhu-flexiv