flexivrobotics/flexiv_rdk #82: [BUG] URDF difference of 4 centimeters for Rizon4

peteflorence, 2024-10-03:
**Version information**
- RDK: v 1.4
- Robot software: 3.6
- OS: Ubuntu 22.04

**Describe the bug**
There are two issues:

1. In [this commit](https://github.com/flexivrobotics/flexiv_rdk/commit/90dfff1e2c7a35aa3266ed62837864f6c3714892), the Rizon4 URDF (`flexiv_rizon4_kinematics.urdf`) changed by >4 centimeters for the end-effector pose (see [script to visualize](https://gist.github.com/peteflorence/b904a82d6240b85ccb1863922804191d))
2. Neither the new URDF nor the old URDF match are able to match the TCP pose reported by the robot.

For problem #2 above, are there hidden calibration parameters (i.e. a per-robot specific URDF) that we could have access to?  Or is there some other cause for the discrepancy?

For problem #1 above, here is the visualization from the script.  I also changed the alpha channel of the robot so that you can more easily see them overlaid

[image]

The specific line change that causes the difference is [here](https://github.com/flexivrobotics/flexiv_rdk/commit/90dfff1e2c7a35aa3266ed62837864f6c3714892#diff-974badbcdbc010ba4ac9663df45e80962bdfc864452acf9b78859d742efbbe00R51).

Is the new URDF or the old URDF closer to correct?  How do we match the TCP pose reported by the robot?

**Steps to reproduce**
1. Grab [this script](https://gist.github.com/peteflorence/b904a82d6240b85ccb1863922804191d)
2. `pip install pybullet`
3. get both the old and new URDFs into the right paths
3. Run the script

**Expected behavior**
URDF should match the robot, i.e. we should be able to take the measured joints and the URDF and calculate the same TCP pose reported by the robot.

**Screenshots**
See screenshot above.

pzhu-flexiv, 2024-10-03:
@peteflorence This URDF is for newly produced Rizon 4 with updated kinematics values. We will add a new API in RDK v1.5 to export the actual URDF used in the control box, but for now, the easiest solution for you is to git checkout to the commit before this URDF change and copy the old URDF out and use that.

pzhu-flexiv, 2024-10-04:
@peteflorence Please advise if this is what you want.
https://github.com/flexivrobotics/flexiv_rdk/commit/7d16e22d2e7463f9501c47453f5443657c82e303

TomCC7, 2025-02-04:
Hi @pzhu-flexiv . I met the same issue number 2 mentioned by @peteflorence 
> Neither the new URDF nor the old URDF match are able to match the TCP pose reported by the robot.

The issue still happens to me even after using the URDF extracted from the `Model::SyncURDF()`. I've tried moving the robot to different configurations and compare the reported tcp pose with the pose calculated from synced URDF (and we tried multiple FK solvers). All the poses have an error of ~5mm. It seems like your discussion didn't resolve the issue mentioned above so I just wonder how to solve it. Thanks!

Version information:
RDK: v 1.5.1
Robot software: 3.7.1
OS: Ubuntu 22.04

pzhu-flexiv, 2025-02-04:
@TomCC7 Have you configured any tool to the robot?

TomCC7, 2025-02-04:
@pzhu-flexiv No, I'm comparing the pose to flange directly

pzhu-flexiv, 2025-02-04:
@TomCC7 In that case we will need more information to pinpoint the root cause. Could you email your sales representative at Flexiv and ask him/her to provide you with instructions to extract robot logs? When done, please send him/her the **log files together with the URDF** you extracted using `Model::SyncURDF()`, and we'll be able to further investigate. Thanks for your time.

closed 2025-07-09 by pzhu-flexiv