flexivrobotics/flexiv_rdk #85: [FEATURE]

yellownebula10, 2024-11-23:
**How urgent is this feature?**
10

**Is your feature request related to a problem? Please describe.**
We control the robot in remote mode. 
We command the robot to move to different target TCP poses but cannot find a way to limit trajectory taken when transitioning from one pose to another. This would be useful to prevent the robot hitting equipment around it.

**Describe the solution you'd like**
Set joint angles values range via RDK. Joint angle limits would be taken into account when robot IK is calculated. 

**Describe alternatives you've considered**
As short-term solution, is there a way to set joint angles limits in Elements that would be considered when IK is calculated? 

Thanks!

pzhu-flexiv, 2024-11-26:
@yellownebula10 We can add setting joint limits over RDK in v1.6. However, only primitives that actually rely on IK calculation  will take into consideration the joint limits, for example MovePTP and MoveIK. The other primitives or the direct motion-force control modes in RDK do not calculate IK to get target joint positions, they calculate target joint torques based on a different control law instead. That being said, they will still honor the joint limits in a different way, which is the joint limits constraining that prevents joints from going over the limits.

yellownebula10, 2024-12-03:
Hi @pzhu-flexiv,
Thank you for your reply. 
When you say "they will still honor the joint limits in a different way, which is the joint limits constraining that prevents joints from going over the limits" does this work when setting safety joint limits? Is setting safety joint limits sufficient to "constraint" the robot path when moving between target TCP poses?

pzhu-flexiv, 2024-12-03:
@yellownebula10 

1. Yes, setting safety joint limits changes both the safety limit and software limit, which is calculated according to the safety limit ( software_limit = safety_limit - const_offset).
2. Yes, if not forced by externally applied torques, the joints won't move over the set limits, i.e. the range of motion is changed. However, this also means some TCP pose might not be reachable due to reduced range of motion in the joints.

pzhu-flexiv, 2025-01-09:
@yellownebula10 This feature is fulfilled in release v1.6, see https://github.com/flexivrobotics/flexiv_rdk/blob/main/include/flexiv/rdk/safety.hpp

closed 2025-01-09 by pzhu-flexiv