flexivrobotics/flexiv_rdk #88: [FEATURE]Support for Windows Python 3.8 is needed.

Coolintang, 2024-12-09:
**How urgent is this feature?**
On a scale of 1 to 10.
10
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. E.g. I'm always frustrated when [...]

I was using Python 3.8 on Windows for development, but there are many old environments that cannot be adapted to other versions of Python. Please help me make it compatible with Python 3.8 on Windows.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

pzhu-flexiv, 2024-12-13:
@Coolintang Please try branch feature/v1.5.1_add_py38_win

pzhu-flexiv, 2025-01-09:
@Coolintang You can now install flexivrdk v1.6 for Python 3.8 on Windows via
```
python3.8 -m pip install numpy spdlog flexivrdk
```

closed 2025-01-09 by pzhu-flexiv