flexivrobotics/flexiv_rdk #91: [BUG] Undefined reference to spdlog (v1.5)

coolzz27, 2024-12-27:
**Version information**
- RDK: v1.5 & v1.5.1 (previous v0.10)
- Robot software: x
- OS: Ubuntu 20.04 - x86_64

**Describe the bug**
I was using v0.10 before and everything was fine. Then I wanted to update to v1.5. I followed the installation guide on linux and managed to install c++ rdk in ~/rdk_install. But when I tried to build in example folder, it failed at this step "cmake --build . --config Release -j 4". Detailed error information is listed below in Additional Context, and it seems to be caused by undefined reference to spdlog. 

Then I removed ~/rdk_install/ and tried to install v1.5.1, and it still failed at the same command. After that, I reinstalled v0.10 and everything was fine. The difference I found between v0.10 and later versions is the thirdparty libraries, including spdlog. So I guess something was wrong with thirdparty dependency?  (Execution of this command finished correctly: "bash build_and_install_dependencies.sh ~/rdk_install")

**Steps to reproduce**
Exactly follows the installation guide and failed at cmake build.

**Expected behavior**
I can successfully build the example.

**Screenshots**
[image]

**Additional context**
(base) vioplum-hp@vioplum-hp:~/Documents/flexiv_rdk-1.5.1/example/build$ cmake --build . --config Release -j 4
/home/vioplum-hp/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -S/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example -B/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build --check-build-system CMakeFiles/Makefile.cmake 0
/home/vioplum-hp/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E cmake_progress_start /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build/CMakeFiles /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build//CMakeFiles/progress.marks
/usr/bin/make  -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
/usr/bin/make  -f CMakeFiles/basics1_display_robot_states.dir/build.make CMakeFiles/basics1_display_robot_states.dir/depend
/usr/bin/make  -f CMakeFiles/basics2_clear_fault.dir/build.make CMakeFiles/basics2_clear_fault.dir/depend
/usr/bin/make  -f CMakeFiles/basics3_primitive_execution.dir/build.make CMakeFiles/basics3_primitive_execution.dir/depend
/usr/bin/make  -f CMakeFiles/basics4_plan_execution.dir/build.make CMakeFiles/basics4_plan_execution.dir/depend
make[2]: Entering directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
cd /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build && /home/vioplum-hp/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E cmake_depends "Unix Makefiles" /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build/CMakeFiles/basics1_display_robot_states.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
cd /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build && /home/vioplum-hp/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E cmake_depends "Unix Makefiles" /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build/CMakeFiles/basics2_clear_fault.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
cd /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build && /home/vioplum-hp/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E cmake_depends "Unix Makefiles" /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build/CMakeFiles/basics3_primitive_execution.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
cd /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build && /home/vioplum-hp/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E cmake_depends "Unix Makefiles" /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build/CMakeFiles/basics4_plan_execution.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
make[2]: Leaving directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
/usr/bin/make  -f CMakeFiles/basics1_display_robot_states.dir/build.make CMakeFiles/basics1_display_robot_states.dir/build
/usr/bin/make  -f CMakeFiles/basics2_clear_fault.dir/build.make CMakeFiles/basics2_clear_fault.dir/build
make[2]: Leaving directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
/usr/bin/make  -f CMakeFiles/basics3_primitive_execution.dir/build.make CMakeFiles/basics3_primitive_execution.dir/build
make[2]: Leaving directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
/usr/bin/make  -f CMakeFiles/basics4_plan_execution.dir/build.make CMakeFiles/basics4_plan_execution.dir/build
make[2]: Entering directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
make[2]: Entering directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
make[2]: Entering directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
make[2]: Entering directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
[  2%] Building CXX object CMakeFiles/basics2_clear_fault.dir/basics2_clear_fault.cpp.o
[  5%] Building CXX object CMakeFiles/basics1_display_robot_states.dir/basics1_display_robot_states.cpp.o
[  8%] Building CXX object CMakeFiles/basics3_primitive_execution.dir/basics3_primitive_execution.cpp.o
/usr/bin/c++ -DFOONATHAN_MEMORY=1 -DFOONATHAN_MEMORY_VERSION_MAJOR=0 -DFOONATHAN_MEMORY_VERSION_MINOR=7 -DFOONATHAN_MEMORY_VERSION_PATCH=1 -DSPDLOG_COMPILED_LIB -isystem /home/vioplum-hp/rdk_install/include -isystem /usr/include/eigen3 -isystem /opt/ros/foxy/include -isystem /opt/ros/foxy/include/foonathan_memory -O3 -DNDEBUG -std=gnu++17 -Wall -Wextra -pthread -MD -MT CMakeFiles/basics1_display_robot_states.dir/basics1_display_robot_states.cpp.o -MF CMakeFiles/basics1_display_robot_states.dir/basics1_display_robot_states.cpp.o.d -o CMakeFiles/basics1_display_robot_states.dir/basics1_display_robot_states.cpp.o -c /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/basics1_display_robot_states.cpp
/usr/bin/c++ -DFOONATHAN_MEMORY=1 -DFOONATHAN_MEMORY_VERSION_MAJOR=0 -DFOONATHAN_MEMORY_VERSION_MINOR=7 -DFOONATHAN_MEMORY_VERSION_PATCH=1 -DSPDLOG_COMPILED_LIB -isystem /home/vioplum-hp/rdk_install/include -isystem /usr/include/eigen3 -isystem /opt/ros/foxy/include -isystem /opt/ros/foxy/include/foonathan_memory -O3 -DNDEBUG -std=gnu++17 -Wall -Wextra -pthread -MD -MT CMakeFiles/basics2_clear_fault.dir/basics2_clear_fault.cpp.o -MF CMakeFiles/basics2_clear_fault.dir/basics2_clear_fault.cpp.o.d -o CMakeFiles/basics2_clear_fault.dir/basics2_clear_fault.cpp.o -c /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/basics2_clear_fault.cpp
/usr/bin/c++ -DFOONATHAN_MEMORY=1 -DFOONATHAN_MEMORY_VERSION_MAJOR=0 -DFOONATHAN_MEMORY_VERSION_MINOR=7 -DFOONATHAN_MEMORY_VERSION_PATCH=1 -DSPDLOG_COMPILED_LIB -isystem /home/vioplum-hp/rdk_install/include -isystem /usr/include/eigen3 -isystem /opt/ros/foxy/include -isystem /opt/ros/foxy/include/foonathan_memory -O3 -DNDEBUG -std=gnu++17 -Wall -Wextra -pthread -MD -MT CMakeFiles/basics3_primitive_execution.dir/basics3_primitive_execution.cpp.o -MF CMakeFiles/basics3_primitive_execution.dir/basics3_primitive_execution.cpp.o.d -o CMakeFiles/basics3_primitive_execution.dir/basics3_primitive_execution.cpp.o -c /home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/basics3_primitive_execution.cpp
[ 11%] Building CXX object CMakeFiles/basics4_plan_execution.dir/basics4_plan_execution.cpp.o
[... cut ...] t::v10::basic_string_view<char>)' follow
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/basics3_primitive_execution.dir/build.make:109: basics3_primitive_execution] Error 1
make[2]: Leaving directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
make[1]: *** [CMakeFiles/Makefile2:170: CMakeFiles/basics3_primitive_execution.dir/all] Error 2
make[1]: Leaving directory '/home/vioplum-hp/Documents/flexiv_rdk-1.5.1/example/build'
make: *** [Makefile:94: all] Error 2

pzhu-flexiv, 2025-01-02:
@coolzz27 It is likely that the `flexiv_rdk` CMake project is using the spdlog installed in your system path. You can force `flexiv_rdk` CMake project to use the spdlog installed in ~/rdk_install by specifying not only `CMAKE_INSTALL_PREFIX`, but also `CMAKE_PREFIX_PATH` to ~/rdk_install:

```
cmake .. -DCMAKE_INSTALL_PREFIX=~/rdk_install -DCMAKE_PREFIX_PATH=~/rdk_install
```

coolzz27, 2025-01-04:
Thank you! The problem has been solved.

closed 2025-01-04 by coolzz27