flexivrobotics/flexiv_rdk #96: Unable to connect to Rizon10 through RDK

simtanzhih, 2025-02-10:
Hi, 

I am trying to connect to my Rizon 10 using RDK through Windows 10 and Visual Studio Code terminal. I am able to ping the robot using the ip, however when I try to run the example basics1_display_robot_states.py, i encounter an error of unable to connect to robot. 

I have already enabled RDK and is running in Auto (remote). I have also recently just upgraded the firmware from v2.10 to v3.6. Previously when I am on firmware v2.10, I am able to connect via Windows and Visual Studio Code. Please do kindly assist.

Thank you.

pzhu-flexiv, 2025-02-10:
@simtanzhih Please provide a screenshot of your terminal output. Also, please make sure you have selected RDK as in the remote mode page.

[image]

simtanzhih, 2025-02-11:
@pzhu-flexiv  please kindly find attached the screenshot from my terminal. I have selected RDK mode in the Flexiv Elements, and also toggle to Auto(remote). 

Thank you.

[image]

pzhu-flexiv, 2025-02-11:
@simtanzhih Have you rebooted the robot after saving changes made to the remote mode setting?

simtanzhih, 2025-02-11:
@pzhu-flexiv yes. I have rebooted the robot.

pzhu-flexiv, 2025-02-11:
@simtanzhih In that case you might need to disable your firewall or add the RDK program to firewall whitelist as stated in https://github.com/flexivrobotics/flexiv_rdk?tab=readme-ov-file#environment-compatibility

simtanzhih, 2025-02-11:
@pzhu-flexiv May I check, I have already disabled my firewall during the run, do I still have to add the RDK program to whitelist? If so, may I know how should I go about doing it?

Thank you.

[image]

pzhu-flexiv, 2025-02-11:
@simtanzhih So when you first start an RDK program, there will be a firewall pop-up prompt asking you whether to allow the program to have network access, when you click yes, it's added to the whitelist.

In addition, could you check in Flexiv Elements UI what software version your robot is running?

simtanzhih, 2025-02-12:
@pzhu-flexiv I did not remember seeing the firewall pop-up prompt when i first start running the program. 

The software version my robot is running on is v3.6

pzhu-flexiv, 2025-02-12:
@simtanzhih Please double check you are using the correct RDK version that's compatible with your robot software version. The version compatibility is documented in both [RDK manual](https://www.flexiv.com/software/rdk/manual/robot_software_compatibility.html) and [release notes](https://github.com/flexivrobotics/flexiv_rdk/releases).

simtanzhih, 2025-02-14:
@pzhu-flexiv Thank you for your help. I did not realized that I had to import flexivrdk from the folder and just followed the example.

closed 2025-02-14 by simtanzhih