flexivrobotics/flexiv_rdk #99: Fail to connect with FlexivRDK v1.6

Tian-Tin, 2025-03-30:
I used to use flexivrkd of old versions to control the robot arm.

And I am tring to make connection to the robot with **flexivrdk v1.6**, which failed.

I am sure that my robot_sn and network_interface_whitelist are both correct, and everything works when it turns to old versions.

So I need some help to figure out where is probably the problem, and my testing code is as follow:

```python
from absl import app, flags
import flexivrdk

flags.DEFINE_string(
    "robot_sn", 
    "Rizon 4s-123456",  # sn for example
    "Serial number of the Flexiv robot"
)
flags.DEFINE_list(
    "network_ip",
    ["10.42.0.1"],  # ip for example
    "Allowed network interfaces/IPs (comma-separated)"
)
FLAGS = flags.FLAGS

def main(_):
    robot = flexivrdk.Robot(
        robot_sn=FLAGS.robot_sn,
        network_interface_whitelist=FLAGS.network_ip 
    )
    print("Connected successfully")

if __name__ == '__main__':
    app.run(main)
```

The error is like:
```
RuntimeError: [flexiy::rdk::Robotl Failed to connect to [Rizon4-123456], Please check that; 
a) Remote mode (Ethernet) is enabled on the target robot; 
b) Ethernet connection with the target robotis established;
c) The specified serial number of the target robot is correct; 
d) No firewall isactive or this RDk program is whitelisted
```
**I am looking forward to any helpful answers.**

pzhu-flexiv, 2025-03-31:
@Tian-Tin Please provide a screenshot of the actual RDK log printed in the terminal.

Tian-Tin, 2025-04-01:
Thanks for your help! The problem lies in my software's version.

closed 2025-04-01 by Tian-Tin