flexivrobotics/flexiv_rdk #103: Question regarding Multi-Robot support

asatwik218, 2025-04-11:
Is it possible to use the Flexiv RDK to connect to and control multiple robot arms simultaneously from a single system or application on Windows?
Specifically, does the API support multiple robot connection requests, and can it manage concurrent operations across multiple arms?
If this is supported, it would be helpful if you could share any documentation, examples, or known limitations (especially on the Windows platform) regarding this functionality.

pzhu-flexiv, 2025-04-11:
@asatwik218 Each `flexiv::rdk::Robot` instance manages everything needed to connect and control one target robot. Therefore, to connect and control N robots at the same time in the same program, you simply instantiate N `flexiv::rdk::Robot` objects. There's no limitation as long as the robots all have unique serial numbers and your network card can handle the data flow. If one network card's bandwidth is saturated, simply add more network cards (PCIe or USB). And when you instantiate `flexiv::rdk::Robot`, specify the `network_interface_whitelist` constructor input parameter to evenly distribute N instances to share M network cards.

pzhu-flexiv, 2025-04-11:
@asatwik218 A side note, if you are talking about controlling a dual-robot setup, you might want to watch the releases from [Flexiv DRDK](https://github.com/flexivrobotics/flexiv_drdk/tree/release/v1.0), which is currently under active development with a lot of practical features specifically tailored for dual-robot controls.

asatwik218, 2025-04-11:
Thanks for the clarification!

We’re planning a 3-robot setup where the center arm has a camera mounted, and the other two arms are equipped with grippers. Each arm will perform a different task but needs to operate in coordination. 
Just to confirm—are there any known issues or limitations when running such a multi-robot configuration with RDK on Windows?
Also, you mentioned keeping an eye on DRDK. Could you share what features DRDK is expected to offer—especially in terms of multi-robot coordination? Since RDK already supports multiple robots, I am curious how DRDK improves upon it. Will it include native synchronization mechanisms, easier inter-robot communication, or coordinated motion planning features?

pzhu-flexiv, 2025-04-11:
@asatwik218 There's no known issue or limitations when running multiple RDK instances on Windows, given your network card hardware have enough bandwidth for the data flow.

DRDK will have a few releases this quarter, adding the following initial features:

1. Synchronized control of two robots.
2. Self-collision avoidance.
3. Bimanual manipulation primitives.

closed 2025-04-21 by pzhu-flexiv