flexivrobotics/flexiv_rdk #110: [QUESTION] Robot.ExecutivePrimitive and Robot.primitive_states behaviour

asatwik218, 2025-05-12:
Hey,

I had a few questions regarding primitives, ExecutePrimitive() , primitive_states() and their behavior. It would be really helpful if you could provide clarifications or direct me to the relevant resources.

1. What is the expected behavior when the robot reaches joint limits or a singularity during a motion primitive? How can this be programmatically detected and managed through the API?
2. Could you please explain the role of the block_until_started parameter in robot.ExecutePrimitive()? How does it influence the overall control flow of the calling program?
3. If block_until_started is set to false and robot.primitive_states() is called immediately after the ExecutePrimitive function call, what information is typically returned?
4. What is the purpose of the properties parameter in robot.ExecutePrimitive()? Where can I get more information on what properties are available?
5. In some scenarios while executing primitives, faults are visible only on the teach pendant or are logged silently, with no interruption or error thrown in program execution. What strategies can be used to ensure the program is aware of and responds to these issues?
6. During a ForceComp primitive command, if the robot fails to reach the target within 5 seconds after trajectory completion, an error is logged, and the robot enters a fault state. Is there a way to override this behavior to allow more time for the robot to complete its trajectory? If not achievable through primitives, can it be managed through other API functions?

Thanks!

pzhu-flexiv, 2025-05-13:
Hi @asatwik218, that's a lot of questions you've got here.

> 1. What is the expected behavior when the robot reaches joint limits or a singularity during a motion primitive? How can this be programmatically detected and managed through the API?

- Regarding joint limits: the robot software will try to keep the joints from moving over the joint limits by generating stopping torques to the joints. However, if the acceleration is too significant or there's a large enough external force, the stopping force from the robot software will be breached, resulting in a fault. Depending on how severe the joint has moved over the limit, a soft fault (CAT2) is first triggered. If the joint keeps travelling further, a hard fault (CAT0) will be triggered, which leads to engaged joint brakes. This is a safety feature, you can refer to the Rizon robot user manual for more details. You can look up the current safety settings in Flexiv Elements -> Settings -> Safety, or using the `rdk::Safety` API. You can detect if a joint is near the joint limit by checking `RobotStates::q` against `RobotInfo::q_min/q_max` which is the soft limit; or against `SafetyLimits::q_min/q_max` which is the hard limit.
- Regarding singularity: the robot software will try to keep the robot from entering a singular pose by generating stopping force/moment in Cartesian space. You can check if the robot pose is near singularity using `Model::configuration_score()`.

pzhu-flexiv, 2025-05-13:
> 4. What is the purpose of the properties parameter in robot.ExecutePrimitive()? Where can I get more information on what properties are available?

It was a temporary workaround for locking the external axes, it is removed in release v1.7 and replaced by dedicated external axes functions. See the v1.7 release note for details.

pzhu-flexiv, 2025-05-13:
> 5. In some scenarios while executing primitives, faults are visible only on the teach pendant or are logged silently, with no interruption or error thrown in program execution. What strategies can be used to ensure the program is aware of and responds to these issues?

If a fault appears on Elements, then it's also reported to RDK, just need to catch it in your program. Say you have a while loop to block and wait for a running primitive to finish, what you can do is adding `robot.fault()` in the while loop to monitor for any fault occurred during the primitive execution. You can also use `rdk::Robot::event_log()` (v1.7) to check all logged events since instantiation.

pzhu-flexiv, 2025-05-13:
> During a ForceComp primitive command, if the robot fails to reach the target within 5 seconds after trajectory completion, an error is logged, and the robot enters a fault state. Is there a way to [...]

Looking at the [primitive doc](https://primitive.flexiv.com/primitives/en/3.9/rizon4/Advanced%20Force%20Control.html#id7), you can try increasing `targetTolerLevel` or setting it to 0.

Alternatively, you can try the direct Cartesian motion-force control using `NRT_CARTESIAN_MOTION_FORCE` mode.

asatwik218, 2025-05-13:
@pzhu-flexiv Thanks for the replies! It was really helpful.
> > 3. If block_until_started is set to false and robot.primitive_states() is called immediately after the ExecutePrimitive function call, what information is typically returned?
> 
> In that case the function will return before the commanded primitive is loaded into the primitive system and has yet to start execute, then primitive_states() should return the previous primitive's states if there is one, or empty if no prior primitive has been executed yet.

Is there a way to get details about the primitive that is currently running? It would be helpful in some applications if primitive_states() would also return little details (such as name, target, etc.) about the primitive itself.

asatwik218, 2025-05-13:
> > 6. During a ForceComp primitive command, if the robot fails to reach the target within 5 seconds after trajectory completion, an error is logged, and the robot enters a fault state. Is there a way to override this behavior to allow more time for the robot to complete its trajectory? If not achievable through primitives, can it be managed through other API functions?
> 
> Looking at the [primitive doc](https://primitive.flexiv.com/primitives/en/3.9/rizon4/Advanced%20Force%20Control.html#id7), you can try increasing `targetTolerLevel` or setting it to 0.
> 
> Alternatively, you can try the direct Cartesian motion-force control using `NRT_CARTESIAN_MOTION_FORCE` mode.

Thanks! that makes sense. 
Can you give some more details on what the scale of `targetTolerLevel` correspond to physically.

pzhu-flexiv, 2025-05-14:
> @pzhu-flexiv Thanks for the replies! It was really helpful.
>
> Is there a way to get details about the primitive that is currently running? It would be helpful in some applications if [...]

Whatever returned from primitive_states() is all information you can obtain from a running primitive.

pzhu-flexiv, 2025-05-14:
> Thanks! that makes sense.
> Can you give some more details on what the scale of `targetTolerLevel` correspond to physically.

Unfortunately I don't have information of the actual values, you may be able to obtain the numbers from your sales representative.