flexivrobotics/flexiv_rdk #119: [BUG] Python - `plan_list()` Raises "No reply from the robot" on RDK 1.7.0

MaxI6s, 2025-07-21:
### Version Info

* **RDK**: 1.7.0
* **Robot Software**: 3.9.3 (simulation)
* **Operating System**: Ubuntu 22.04

### Urgency

**8 - Blocking.**
This issue is preventing us from migrating our workflow from RDK 1.6.0 to 1.7.0.

### Description

Calling `plan_list()` on a connected simulated robot consistently raises a `RuntimeError` with the message:

```
RuntimeError: No reply from the robot
```

**Expected Behavior:**
The function should return the list of available plans deployed on the robot, as it did in RDK 1.6.0.

### Steps to Reproduce

1. Launch a simulated robot using Flexiv Elements Studio.
2. Establish a connection using the RDK:

   ```python
   rbt = flexivrdk.Robot("Rizon##-######")
   ```
3. Call:

   ```python
   rbt.plan_list()
   ```
4. Observe the error:

   ```
   RuntimeError: No reply from the robot
   ```

### Additional Notes

* This was working as expected with RDK 1.6.0.
* The connection is successfully established (`connected()` returns `True`).
* Other methods like `ExecutePlan()`  do not raise errors.

pzhu-flexiv, 2025-07-21:
@MaxI6s Does both Elements Studio and RDK run inside the same WSL environment?

MaxI6s, 2025-07-21:
@pzhu-flexiv, Both are working properly.

[image]

pzhu-flexiv, 2025-07-21:
@MaxI6s Could you try global_variables() ?

MaxI6s, 2025-07-21:
With Flexiv Elements Studio running natively on a Linux machine and the RDK client running inside WSL2, plan_list() seems to work correctly.

[image]

pzhu-flexiv, 2025-07-22:
@MaxI6s In that case the issue should come from using Elements Studio inside WSL, are you able to continue the development using the working method for now?

MaxI6s, 2025-07-24:
@pzhu-flexiv, it’s a good solution for now, and we can continue developing Elements. However, we are definitely looking for a stable way to run Elements with WSL in the future.
Thanks for your support!

closed 2025-07-29 by pzhu-flexiv